keyboardio/Kaleidoscope #1320: Occasional dropped keyswitch events

gedankenexperimenter, 2022-12-23:
This was originally reported as qukeys getting "stuck" on, and there was much discussion on Discord.  Thanks to @SiriusStarr for the original bug report and help in determining the cause.

The problem is that some plugins that use `KeyAddrEventQueue` to help handle keyswitch events will occasionally drop an event, which can lead to either a keypress not being registered or a key getting "stuck" (because a key release was dropped).  After debugging with a plugin that records an event log over serial, we discovered that all the dropped events had the event ID `127`, which is the maximum value of an `int8_t`.  It appears that I mistakenly assumed that the compiler wouldn't implicitly promote those integers to 16-bit ones before doing a comparison in the `shouldAbort()` function.  It's still a mystery why this ever worked right on the AVR processors, but it definitely failed on GD32.

I'm going to try to reproduce it in the simulator with a testcase.  Whether or not it's reproducible there, I'll submit a PR to fix the issue.

gedankenexperimenter, 2022-12-23:
The simulator seems to be dead in the water on Mac OS 13.1, so I'll just go ahead and try submitting the fix without a test.

obra, 2022-12-23:
That’s an amazing catch.

tlyu, 2022-12-23:
What's puzzling to me is that this should be failing on AVR as well. I just disassembled an Atreus firmware compiled from `master`, and it shows the integer promotion happening in `shouldAbort`.

```
   48a0:       24 81           ldd     r18, Z+4        ; 0x04
  ///
  /// Returns `true` if the input event is newer than the latest event the
  /// tracker has seen.  If it is newer, also update the tracker to reflect the
  /// new event ID.
  bool isNew(const KeyEvent &event) {
    KeyEventId offset = event.id() - last_id_seen_;
    48a2:       8d a5           ldd     r24, Y+45       ; 0x2d
    48a4:       92 2f           mov     r25, r18
    48a6:       98 1b           sub     r25, r24
```

The above does a signed 8-bit subtraction, as expected.

```
    if (offset > 0) {
    48a8:       19 16           cp      r1, r25
    48aa:       2c f0           brlt    .+10            ; 0x48b6 <_ZN12kaleidosc
ope6plugin6Qukeys16onKeyswitchEventERNS_8KeyEventE+0x22>
    return KeyEvent{addr(i), state, Key_Undefined, id(i)};
  }

  // Only call this after `EventTracker::shouldIgnore()` returns `true`.
  bool shouldAbort(const KeyEvent &event) const {
    return (length_ != 0) && (event.id() - event_ids_[0] >= 0);
    48ac:       8b 81           ldd     r24, Y+3        ; 0x03
    48ae:       81 11           cpse    r24, r1
    48b0:       1c c0           rjmp    .+56            ; 0x48ea <_ZN12kaleidoscope6plugin6Qukeys16onKeyswitchEventERNS_8KeyEventE+0x56>
```

There's some confusing-looking cross-jumping because of the high optimization levels involved.

```
    48ea:       02 2e           mov     r0, r18
    48ec:       00 0c           add     r0, r0
    48ee:       33 0b           sbc     r19, r19
    48f0:       8c 81           ldd     r24, Y+4        ; 0x04
    48f2:       28 1b           sub     r18, r24
```

The above does the 8-bit signed subtraction, but the extra stuff surrounding it compensates so that the sign comparison is correct for the result after integer promotion.

```
    48f4:       31 09           sbc     r19, r1
    48f6:       87 fd           sbrc    r24, 7
    48f8:       33 95           inc     r19
    48fa:       37 fd           sbrc    r19, 7
    48fc:       da cf           rjmp    .-76            ; 0x48b2 <_ZN12kaleidoscope6plugin6Qukeys16onKeyswitchEventERNS_8KeyEventE+0x1e>
```

More cross-jumping.

```
    return EventHandlerResult::OK;
    48b2:       80 e0           ldi     r24, 0x00       ; 0
    48b4:       25 c0           rjmp    .+74            ; 0x4900 <_ZN12kaleidosc
ope6plugin6Qukeys16onKeyswitchEventERNS_8KeyEventE+0x6c>
```

```
      return EventHandlerResult::ABORT;
    48fe:       82 e0           ldi     r24, 0x02       ; 2
}
    4900:       df 91           pop     r29
    4902:       cf 91           pop     r28
    4904:       1f 91           pop     r17
    4906:       0f 91           pop     r16
    4908:       08 95           ret
```

gedankenexperimenter, 2022-12-23:
I'm back to thinking this doesn't make sense again.  The call to `KeyEventTracker.shouldIgnore()` should prevent `KeyAddrEventQueue.shouldAbort()` from getting called at all.  I've tried to reproduce this in the simulator, so far without success.

tlyu, 2022-12-23:
It seems to me like `shouldAbort` is only called if `shouldIgnore` returns true.
https://github.com/keyboardio/Kaleidoscope/blob/506c780f780148f5cd2472990af4de32debda98d/plugins/Kaleidoscope-Qukeys/src/kaleidoscope/plugin/Qukeys.cpp#L45-L48

gedankenexperimenter, 2022-12-23:
Okay, forget my last comment; I've been so confused by this bug that I lost track of the fact that I was running the test in the simulator without first backing out the fix that I committed once I gave up on getting the simulator working.  I can reproduce it now.

gedankenexperimenter, 2022-12-23:
> It seems to me like `shouldAbort` is only called if `shouldIgnore` returns true.

You are, once again, absolutely correct.  Thanks for providing some clear thinking (repeatedly) when my brain was not up to the task.

gedankenexperimenter, 2022-12-23:
I'll put together the PR and submit it later this evening.  It will be a big relief to get this one behind me.

SiriusStarr, 2022-12-24:
Thanks for your tireless work on fixing my, and weirdly only my, issue. :laughing:

commit 27bd385c (2022-12-24): Fix event dropping when id overflow occurs

The compiler promotes our signed 8-bit integer to a 16-bit integer, at least on
some processors.  This causes the wrong result for `shouldAbort()` when the new
event's id is -128, and the one at the head of the queue is 127.

Fixes #1320
closed 2022-12-24 by obra