keyboardio/Kaleidoscope #1324: PrefixLayer does not work with more than one prefix layer.

LemonBreezes, 2023-01-12:
With this config snippet, only the first prefix layer works:
[image]

[image]
Manually editing the `prefix_layers_length_` in the highlighsed line of code to 3 fixes this issue for me so I believe the length is not being calculated correctly.

I'm using a Model 100 and I'm on the latest master branch of Kaleidoscope.

gedankenexperimenter, 2023-01-13:
There's an error in `setPrefixLayers()`, which I'm pretty sure I can fix with a template.

gedankenexperimenter, 2023-01-13:
I'm pretty sure I've got this one worked out; I just need to write a testcase for multiple prefix layers.

PR #1332 (2023-03-15): Fix defining multiple prefix layers
The function `setPrefixLayers()` had a flaw: its parameter was a pointer to an array, without a length, resulting in the `prefix_layers_length_` variable not being set correctly.  It was meant to compute the length by dividing the size of the whole user-defined `prefix_layers` array by the size of a single element, but was instead using the size of a pointer in place of the size of the whole array.  The included testcase needs three prefix layers in order to trigger the bug because the size of a pointer in the simulator is 8 bytes.  Using a template function [... cut]
closed 2023-03-15 by obra