keyboardio/Kaleidoscope #1329: Model 100 LED regression: only some LEDs light on boot

iliana, 2023-02-18:
**Describe the bug**
In the default sketch, the first LED effect is LEDOff. On [my firmware sketch](https://github.com/iliana/Kaleidoscope/blob/011025748fea5a334dc97968d1f872470cd21b0f/iliana/Model100/Model100.ino), I enable a solid color LED effect immediately (and have no other LED effects).

Somewhere in [`becf816...c1acb3b`](https://github.com/keyboardio/Kaleidoscope/compare/becf816...c1acb3b) (this comparison looks confusing due to the LEDEffect-DigitalRain repo merge) seems to introduce a regression for this firmware. On power up, some or none of the LEDs light. I can hit my LED key (bound to `Key_LEDToggle` because I have only one effect) twice, and all LEDs light as expected.

When some LEDs do light, there are patterns. Generally it is the right 2 columns of the right half, and only the LED key on the left half. Sometimes we also get the tab and left Fn key.

7e2a7672536848dc25fc5e32f4f665a0e8d3e337 looks potentially related? I am not sure.

**To Reproduce**
1. Flash [my sketch](https://github.com/iliana/Kaleidoscope/blob/011025748fea5a334dc97968d1f872470cd21b0f/iliana/Model100/Model100.ino).

**Expected behavior**
I expected all the LEDs to light.

**Environment (please complete the following information):**
 - OS: macOS
 - Version: 12.6
 - Device: Keyboardio Model 100

iliana, 2023-02-18:
Adding this to my `setup()` function reliably works around this issue:

```cpp
LEDControl.disable();
delay(100);
LEDControl.enable();
```

obra, 2023-02-21:
@tlyu Do you think this might be related to some of your recent rework on startup?

tlyu, 2023-02-21:
> @tlyu Do you think this might be related to some of your recent rework on startup?

Yeah, it's possible. I'll try to reproduce, and I can try adding a poll for both ATtinys to start responding before exiting the Model100 setup (with a reasonable timeout in case someone is running with the right half disconnected).

tlyu, 2023-02-22:
I can't reproduce it by setting my default LED mode to one of the solid colors on a default Model 100 sketch. I'm going to assume that the custom sketch speeds things up enough that the solid color mode writes the LEDs before the ATtinys have booted up.

iliana, 2023-02-22:
That's my assumption as well. (Even with the same firmware build on my sketch, different numbers of LEDs come up on the left half on each boot, so I think the 100ms workaround I added is probably overkill.)