keyboardio/Kaleidoscope #1330: Recent versions of arduino-cli appear to require a sketch.yaml instead of sketch.json

iliana, 2023-02-18:
When getting my keyboard flashing environment up and running again, I ran into this issue:

```
$ arduino-cli version
arduino-cli  Version: 0.30.0 Commit: 83700ca2 Date: 2023-02-08T13:35:24Z
$ make compile
Using Kaleidoscope from /Users/iliana/git/Kaleidoscope
Building in quiet mode. For a lot more information, add 'VERBOSE=1' to the beginning of your call to /Library/Developer/CommandLineTools/usr/bin/make
Missing FQBN (Fully Qualified Board Name)
*************************************************************** 

Arduino couldn't figure out what kind of device this sketch 
is for. Usually, Arduino looks in a file called `sketch.json` 
to figure this out. 

I'm unable to detect your keyboard, you may need to manually 
edit your `sketch.json` file or run 

ARDUINO_DIRECTORIES_USER=/Users/iliana/git/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/Users/iliana/git/Kaleidoscope/.arduino/data /opt/homebrew/bin/arduino-cli board attach 

manually, specifying the FQBN for your keyboard. 

*************************************************************** 
/Users/iliana/git/Kaleidoscope/etc/makefiles/sketch.mk:93: *** .  Stop.
```

Running the suggested `board attach` command generates a new sketch.yaml:

```
$ ARDUINO_DIRECTORIES_USER=/Users/iliana/git/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/Users/iliana/git/Kaleidoscope/.arduino/data /opt/homebrew/bin/arduino-cli board attach -b keyboardio:gd32:keyboardio_model_100
Default port set to: <nil>
Default FQBN set to: keyboardio:gd32:keyboardio_model_100

$ cat sketch.yaml 
default_fqbn: keyboardio:gd32:keyboardio_model_100
```

I am not sure what versions of arduino-cli introduced sketch.yaml and subsequently stopped reading sketch.json, since it's been a while since I've attempted to compile my firmware and have run `brew upgrade` several times since then.

tlyu, 2023-02-21:
I recently ran into this myself after upgrading `arduino-cli` through Homebrew. This behavior seems to be new in `arduino-cli` 0.30. compare https://arduino.github.io/arduino-cli/0.29/sketch-specification/#sketchjson and https://arduino.github.io/arduino-cli/0.30/sketch-specification/#sketchjson

obra, 2023-02-21:
As soon as the test suite finishes running, I'll merge in #1331, which
fixes this.

PR #1331 (2023-02-21): Arduino now requires sketch.yaml files instead of sketch.json files.
Fixes #1330 1330
closed 2023-02-21 by obra