keyboardio/Kaleidoscope #1338: EEPROM taking precedence over PROGMEM for layers and colormaps

a-dma, 2023-03-16:
**Describe the bug**
When defining layers or colormaps (through the Colormap plugin) in code (i.e., not using Chrysalis) subsequent changes after the first time the firmware is flashed, are ignored.

It looks as if the layers and colormaps in PROGMEM are copied to EEPROM only if they don't already exist there.

I suspect this is some sort of regression, but perhaps it's down to some nuances I haven't understood and ultimately it's intended behavior.

**To Reproduce**
A minimal example would be:
- create a sketch with one layer (but the same applies for colormaps)
- compile and flash the firmware (call this `v1`)
- verify that the keymap works as expected
- update the layer keymap by making a change, any change
- compile and flash the new firmware (call this `v2`)
- the kemap is still the same as `v1`

**Expected behavior**
I would expect the keymap to be `v2` instead of `v1`. 

**Environment (please complete the following information):**
 - OS: Linux
 - Version v1.99.7-101-g4381daba aka commit `4381daba80993c68c32c2b8def621e942e6ec603`
 - Device Keyboardio Model 01

**Additional context**
I've troubleshooted this down to EEPROM taking precedence over PROGMEM.

If I manually send the `keymap.onlyCustom` command, I can see it returns `1` and if I set that to `0` then the layers and colormaps that were defined in code start working. This is a possible compromise, but a) I need to keep a script around to manually send focus commands and b) Chrysalis (which I don't use but I'd like to keep as an option) doesn't support `keymap.onlyCustom` off and produces a warning dialog.

Additionally, if I manually send the `eeprom.erase` command and re-flash the same code, then everything works as it should.

In essence, it feels like the layers and colormaps from PROGMEM are copied into EEPROM if the corresponding EEPROM slice is empty, I would expect that to happen every time the firmware is flashed. Should flashing a new firmware erase EEPROM, perhaps?

I somehow managed to work around this in code for the colormaps by using `DefaultColormap.install()` rather than `DefaultColormap.setup()`, but I'm starting to wonder whether this is a viable solution: if I understand correctly what is happening, the colormap EEPROM slice is rewritten every time that the keyboard is powered on. If that's the case, it's probably not great long term. Could you please confirm if this is a bad idea?

tl;dr questions:
- is it OK to call `DefaultColormap.install()` inside `setup()`
- how can I copy the content of PROGMEM into EEPROM whenever I flash a new firmware without turning on `keymap.onlyCustom`?

Thanks in advance for the help!

--
One final item, the documentation of `keymap.onlyCustom` is a bit inconsistent. It's mentioned both [here](https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-EEPROM-Keymap.html) and [here](https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-EEPROM-Keymap.html), but the plugin method is referred to as both `EEPROMSettings.use_eeprom_layers_only` and `ignoreHardcodedLayers`, whereas the focus command is referred to as both `keymap.useCustom` and `keymap.onlyCystom`. I believe that in both cases the second name is the correct one.

Let me know if I should open a dedicated issue for this.

obra, 2023-03-16:
By design, layers in EEPROM take precedence over hardcoded layers. If
you're looking to reset the eeprom on flash, you can use the focus command
"eeprom.erase" to wipe your eeprom.

Alternatively, if you're not currently using Chrysalis, why not just remove
support for EEPROM keymaps?

obra, 2023-03-16:
> First of all, thanks for the reply.
> Thanks for clarifying that EEPROM layers are supposed to take precedence. I think what I find surprising is that if the EEPROM is empty, then programmed layers are copied there, but only the first time. I would expect that to either happen every time, or not at all.
> Is there a way to make that behavior happen every time? Alternatively, I assume that when you say to send the eeprom.erase command you mean as a manual additional step? I can do that, but perhaps there's a built-in way already?
> That's correct, I'm not using Chrysalis right now, but I wouldn't mind being able to. Of course I can change things down the line "if and when".

The idea is that Chrysalis should never overwrite anything in eeprom. customizing your layouts in code is behavior that is important for us to support, but not the default for most users. If you want to make the behavior, you describe your default, a little script that echoes “eeprom.erase” to the keyboard over serial after flashing is your best bet.

a-dma, 2023-03-20:
Fair. I don't know what the process looks like with Chrysalis as I've used that tool only once or twice as a visualization/debug aid, but in my mind what you describe can be addressed by a backup+restore+verify cycle as part of the firmware upgrade. That would also have the added benefit of leaving the user with a backup of their configuration and allowing to decouple the on-disk storage format and the internal one should the latter ever need to change between firmware versions (which I understand might be a solution to a non-issue, but it'd be free in this case).