keyboardio/Kaleidoscope #1350: CharShift doesn't work with Macros

mkan88, 2023-07-08:
**Describe the bug**
If a macro is defined as one of the keys of CharShift (either non-shifted, shifted) then no output will be shown for the macro key.

**To Reproduce**
Steps to reproduce the behavior:
Add a `KeyPair` involving a macro to `CS_KEYS`, then add it to the keymap (e.g. `CS(0)`)
```
CS_KEYS(
    kaleidoscope::plugin::CharShift::KeyPair(Key_1, M(MACRO_ANY)),
)
```
where M is the macro for macros.

**Expected behavior**
The macro should be executed as usual.

**Environment (please complete the following information):**
 - OS: [Fedora]
 - Version [37]
 - Device [Keyboardio Model 01]

**Additional context**
I was trying to add an APL layer to my keyboard, which is why I'm attempting to do it this way rather than having the shift key on the APL layer temporarily shift to another layer. 

I'm happy to contribute to fixing the bug, but I'm not sure where to start. Any pointers would be appreciated.

Thank you for the team's hard work!

gedankenexperimenter, 2023-07-17:
I'm guessing that you have `CharShift` listed before `Macros` in `KALEIDOSCOPE_INIT_PLUGINS()`, and if you change the order, it would work as expected.  At the very least, this should be noted in the docs for plugins that don't restart event processing, and maybe we should consider having CharShift restart event processing, and use some mechanism to try to keep from getting caught in an infinite loop.

The general problem is that, if you've got two plugins, each of which converts a key event to a type handled by the other, the later one would need to restart the event processing in order for the earlier one to handle it.  In this regard, Macros should probably be placed very late in the plugin order, because it usually restarts event processing, and also rarely needs to generate events handled by other plugins.

mkan88, 2023-08-03:
`CharShift` is the last entry in `KALEIDOSCOPE_INIT_PLUGINS()`, so that's unlikely to be the issue. I'll try to to create a minimal reproducible example.

gedankenexperimenter, 2023-08-30:
Please forgive me; I wrote "before" when I meant to write "after".  For this to work with the current plugins, your CharShift plugin needs to come first, then Macros.

mkan88, 2023-09-18:
That seems to have worked. Thanks for the help!

closed 2023-09-18 by mkan88