keyboardio/Kaleidoscope #1351: PrefixLayer doesn't work on my Model 01

iliana, 2023-07-11:
**Describe the bug**
The [PrefixLayer plugin](https://github.com/keyboardio/Kaleidoscope/tree/master/plugins/Kaleidoscope-PrefixLayer) doesn't seem to work on my Model 01, while the same sketch builds and works as expected on my Model 100. (I'm trying to bring my Model 01s up to my current sketch so I can use them at other workstations.)

**To Reproduce**
Steps to reproduce the behavior:
1. Build and upload [the PrefixLayer example sketch](https://github.com/keyboardio/Kaleidoscope/tree/master/examples/Keystrokes/PrefixLayer) to a Model 01.
2. Press and release Butterfly+L while monitoring key inputs.

**Expected behavior**
The expected behavior of this sketch is to see Ctrl+B pressed and released, then the right arrow key pressed and released.

**Actual behavior**
Only the right arrow key is pressed and released. (This seems to imply that the layer transition is being handled fine, but the usual events that should fire within the PrefixLayer plugin are not.)

**Environment (please complete the following information):**
 - OS: macOS; Arch Linux
 - Version: 12.6.x; N/A
 - Device Keyboardio Model 01

**Additional context**
When initially developing this plugin I was developing against a Model 100. I'm unsure how best to go about debugging this issue; I'm not sure what differences in the hardware or firmware between the Model 01 and Model 100 might be causing this.

iliana, 2023-07-11:
Further context: the last (out-of-tree) version of this plugin that I'm aware of working on the Model 01 can be found at https://github.com/iliana/Kaleidoscope-PrefixLayer/blob/a4e40f6b01047903b3225ceae7f0ccf101ed1284/src/Kaleidoscope/PrefixLayer.cpp.

obra, 2023-07-11:
Does the PrefixLayers .ktest file exercise this specific case? (That runs
on a simulated Model 01, but there's a chance that there's a hardware quirk
we're missing)

iliana, 2023-07-11:
Yeah this is that case: https://github.com/keyboardio/Kaleidoscope/blob/90abb32cc9eee25fa32560528e4b7b52d6441c6d/tests/plugins/PrefixLayer/basic/test.ktest#L37-L62

"Prefix layer passthrough" is also a case that does not work on hardware.

iliana, 2024-02-07:
I finally sat down to print-debug this, and my friend and AVR-knower sitting behind me pointed out that we define the prefix layers as PROGMEM:

https://github.com/keyboardio/Kaleidoscope/blob/1780dc9c8076374cc733dfb9ba7626840958b7ca/examples/Keystrokes/PrefixLayer/PrefixLayer.ino#L20-L23

Presumably in the simulated Model 01 (an AVR), the particular quirk of PROGMEM being completely different address space that requires different instructions to read from it is not simulated. This also does not matter on the Model 100, an ARM device with unified memory.

Reading the layer out of the prefix layers array here _always_ returns zero on Model 01 hardware, resulting in everything breaking unexpectedly:

https://github.com/keyboardio/Kaleidoscope/blob/1780dc9c8076374cc733dfb9ba7626840958b7ca/plugins/Kaleidoscope-PrefixLayer/src/kaleidoscope/plugin/PrefixLayer.cpp#L47

I am going to operate on the assumption that we do want users to declare this array as PROGMEM, and fix up the plugin to perform reads properly.

PR #1386 (2024-02-07): PrefixLayer: read layers from PROGMEM
The documentation and examples for the plugin specify that the prefix layers are PROGMEM, but we didn't actually read from PROGMEM, resulting in the plugin working improperly on AVR hardware.

Tested on a Model 01.

Fixes #1351.
commit 738b4f69 (2024-02-07): PrefixLayer: read layers from PROGMEM

The documentation and examples for the plugin specify that the prefix
layers are PROGMEM, but we didn't actually read from PROGMEM, resulting
in the plugin working improperly on AVR hardware.

Tested on a Model 01.

Fixes #1351.
closed 2024-02-07 by obra