keyboardio/Kaleidoscope #1354: Colormap example does not work

nicos68, 2023-08-22:
The Colormap.ino source code that can be found under `examples/LEDs/Colormap` does not behave as expected.

Steps to reproduce the behavior:
 - Compile the source code of `Colormap.ino`
 - Flash the binary to the keyboard

**Expected behavior**
 - The colors defined in the Colormap.ino source code are applied to the keyboard's LEDs

**Actual behavior**
 - Seemingly random colors are applied to the keyboard's LEDs.
 - When running Chrysalis the colors displayed by the tool acutally match the keyboard LEDs' colors:
[image]
 - after a few seconds the LEDs of the keyboard's right half turn off.
 - If I use chrysalis to change a led color, it actually gets applied to the keyboard's LEDs

**Environment (please complete the following information):**
 - OS: Fedora
 - Version 38
 - Device Keyboardio Model 01

**Additional context**
When doing a factory reset, no matter what color I set using Chrysalis, it never gets applied to the keyboard's LEDs

nicos68, 2024-03-24:
So it looks like I have to run the backup script after flashing for the colormap plugin to work, that’s confusing but well, I found a solution.

obra, 2024-03-25:
I'm glad you got it sorted out. I'm not quite sure what caused the weird
behaviour you're seeing. If you get it stuck again, please let us know.

Can you tell me which backup script you were using and whether you flashed
with Chrysalis or with 'make flash' from the commandline?

Best,
jesse

nicos68, 2024-03-25:
- I use kaleidoscope’s eeprom-backup script
- I flash my keyboard
  - Using Chrysalis when I make simple keymap changes
  - Using [arduino-cli](https://arduino.github.io/arduino-cli/0.35/) when I make code changes to the firmware code. See the flash (and compile) recipe in [my justfile](https://codeberg.org/nise/Model01-Firmware/src/commit/f93e862a8d7579bbdb8c4df5040f757102e58839/justfile#L13)

obra, 2024-03-25:
indeed, if you are storing things in the on devices storage, you should
either be using Chrysalis to do your flashing or always be doing an EEPROM
back up and restore on the the two sides flashing

nicos68, 2024-03-25:
Well, I would expect things to fix themselves when I execute the restore script, but that's not what happens :
 * I added the colormap plugin to my firmware code
 * I compiled the firmware
 * I flashed the new firmware with colormap and my color settings => The random colors showed up and my colormap settgins were ignored
 * I ran the backup scrip => then immidiately my colormap settings were applied to the keyboard. I did not run the restore script.

obra, 2024-03-25:
That’s odd. You *should* need to run the backup before flashing

nicos68, 2024-03-26:
It works like that everytime now. If I change the colormap in my firmware, then compile and flash, the color change is ignored. Then when I run the backup script the colormap change is correctly applied.

closed 2024-11-23 by obra