keyboardio/Kaleidoscope #1393: Sticky modifier doesn't work with the same key on a different layer.

NabiYang, 2024-02-22:
**Describe the bug**
If a sticky modifier is activated from a layer, then that layer is released, pressing the same key doesn't work.

**To Reproduce**
1. Place Ctrl on a shifted layer on the same key that types 'A', for example. Make it Sticky.
2. Save that layout to the board.
3. In a text editor where Ctrl+A would select all text, type some text.
4. Hold the layer and tap the sticky Ctrl.
5. Release the layer and tap 'A'--the same key that had the sticky Ctrl.
6. ---> It doesn't select all text.

**Expected behavior**
I expect that the sticky Ctrl would apply to A. The sticky modifier does apply correctly so long as the key isn't the same. I tested this with Ctrl and Win.

**Environment (please complete the following information):**
 - OS: Windows
 - Version 11
 - Device Atreus, Model 100

**Additional context**
This was reported in Discord here: https://discord.com/channels/492408953041321984/761681781447196693/1210253884824625202

obra, 2024-02-23:
Hm. I can intuit why that might be happening. For an "active" key,
Kaleidoscope needs to track it by where on the keyboard it is, which means
that once that sticky Ctrl is active, we don't have a slot to stick the A
into.

riantsoa, 2024-02-29:
> Hm. I can intuit why that might be happening. For an "active" key, Kaleidoscope needs to track it by where on the keyboard it is, which means that once that sticky Ctrl is active, we don't have a slot to stick the A into.
> […](#)

On QMK and ZMK it is working fine.