keyboardio/Kaleidoscope #1421: Hardware test mode documentation

EvyBongers, 2024-05-05:
I looked and I couldn't find...

It would be really useful to have documentation on the hardware test mode, because right now it's nowhere to be found and therefore not at all usable unless @obra is available to help with that.

obra, 2024-05-06:
Good catch.

I'm sure I've written that documentation multiple times, too. https://github.com/keyboardio/Kaleidoscope/wiki/Hardware-Test-Mode#using-hardware-test-mode is old and insufficient, but what I'm going to crib from as I write it properly.

obra, 2024-05-23:
There's now a page at https://shop.keyboard.io/pages/model-100-keyswitch-test-mode linked from https://shop.keyboard.io/pages/support 

I'm very happy to take suggestions on making the page better.

closed 2024-05-23 by obra