keyboardio/Kaleidoscope #1425: ColormapOverlay overlays don't seem to apply to transparent keys

obra, 2024-05-23:
I was a little surprised, when testing out the replacement for the Numpad plugin, that setting an overlay on a key that didn't have anything defined on the numpad layer...just didn't light up the key. I'd expect that if I've explicitly defined an overlay, it wouldn't skip lighting up just because there was a transparent key on the keymap.

@EvyBongers - is this an intentional design? I'd love to understand the design intent better.

EvyBongers, 2024-05-24:
I don't believe that was intentional. This plugin is one of my first contributions, so I may have simply overlooked something. Will have a look this weekend

EvyBongers, 2024-05-26:
I don't do anything in particular to prevent the led from being set if the key isn't assigned. Basically, if an overlay is defined for the given key address on that layer, `::LEDControl.setCrgbAt` is called to set the color for that particular key.
https://github.com/keyboardio/Kaleidoscope/blob/2c280ddda5c6b9f8c029e3509ce3d9f2cd24f6dd/plugins/Kaleidoscope-Colormap-Overlay/src/kaleidoscope/plugin/Colormap-Overlay.cpp#L61-L67
I think the error I made is that I used `Layer.lookupActiveLayer` rather than `Layer.mostRecent` to find the current layer. 
https://github.com/keyboardio/Kaleidoscope/blob/2c280ddda5c6b9f8c029e3509ce3d9f2cd24f6dd/plugins/Kaleidoscope-Colormap-Overlay/src/kaleidoscope/plugin/Colormap-Overlay.cpp#L37-L51

Does that make sense to you?

EvyBongers, 2024-05-26:
Now I realize that this behavior may have been intentional, even it if is counterintuitive... My use-case for the overlay plugin is to highlight my layer switches. The way it's currently set up supports lighting up keys in lower layers, provided that higher layers don't have a key assigned (i.e. transparent). I suppose it should be easy enough to add additional logic to check for an assigned color overlay at the top-most layer before falling through to lower layers. I could also add a toggle to enable/disable the option to check for overlays on lower layers.

PR #1432 (2024-06-18): Have ColormapOverlay support blank keys
Apply color overlays regardless of the value of the key in the current (top most) layer.
This change aims to maintain backwards compatibility to apply color overlays on lower layers, provided that the top most layer does not specify a color for a key.

Due to limitations of the API, there is no guarantee that the color applied matches the highest layer matching the key except for the top most layer.

Fixes #1425
closed 2024-06-18 by obra