keyboardio/Kaleidoscope #1429: Warning when running the test suite - duplicate member name

obra, 2024-05-28:
When running `make simulator-tests` on macos, I'm seeing this warning.  I'm wondering if the right solution is a fix to build tooling or renaming the source file for the tinyusb hid driver. Thoughts, @tlyu?

```/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning duplicate member name 'AbsoluteMouse.cpp.o' from '/Users/jesse/Documents/GitHub/keyboardio/firmware/Kaleidoscope/_build/keyboardio_virtual_model01/kaleidoscope/build/1760398448-release-cleared.ino/libraries/Kaleidoscope/kaleidoscope/driver/hid/tinyusb/AbsoluteMouse.cpp.o(AbsoluteMouse.cpp.o)' and '/Users/jesse/Documents/GitHub/keyboardio/firmware/Kaleidoscope/_build/keyboardio_virtual_model01/kaleidoscope/build/1760398448-release-cleared.ino/libraries/KeyboardioHID/KeyboardioHID.a(AbsoluteMouse.cpp.o)'
``

tlyu, 2024-05-28:
Interesting, I thought `find-filename-conflicts` would have picked up on it. I already did a rename for some WIP stuff because CI caught it.

Is that the only conflict, or are there others?

tlyu, 2024-05-28:
Ah, the answer is apparently that `find-filename-conflicts` is buggy, and only looks for conflicts in each given directory tree individually, instead of gathering all the filenames in all the specified trees.

tlyu, 2024-05-28:
Current list, using fixed script:

```
bin/find-filename-conflicts.py src plugins/*
Searching for conflicting filenames...
Conflict found for file name 'AbsoluteMouse.cpp':
  -> src/kaleidoscope/driver/hid/tinyusb/AbsoluteMouse.cpp
  -> plugins/KeyboardioHID/src/MultiReport/AbsoluteMouse.cpp
Conflict found for file name 'twi.c':
  -> plugins/Kaleidoscope-Hardware-Keyboardio-Imago/src/kaleidoscope/device/keyboardio/twi.c
  -> plugins/Kaleidoscope-Hardware-Keyboardio-Model01/src/kaleidoscope/device/keyboardio/twi.c
gmake: *** [Makefile:102: find-filename-conflicts] Error 2
```

The `twi.c` conflict might only matter for the cache warmer sketch, and macOS `libtool` already works around it (with a warning).

The other one doesn't seem to cause a problem in practice, possibly because `avr-ar` includes pathnames in the archive when specified.

PR #1430 (2024-05-28): fix find-filename-conflicts
Fix some bugs in `find-filename-conflicts`: it was checking each subtree from its command line independently, rather that combining all of the basenames in the merged trees.

Clean up the existing code somewhat.

Also rename existing filename conflicts.

Fixes #1429.
closed 2024-05-28 by obra