keyboardio/Kaleidoscope #1438: multiple ShiftToLayer keys should behave like one

mrwonko, 2024-10-30:
**Describe the bug**
When I press both of my "fun" keys and release one, the layer shift is undone.

**To Reproduce**
Bind two keys to Shift to Layer 2. Hold both of them. Release one.

**Expected behavior**
While at least one of the Shift to Layer keys for the same layer is held, I want to stay on that layer. The shift should only be undone once I release all the shift keys, similarly to how holding both "shift" keys on a keyboard and releasing one still lets me write uppercase.

**Environment (please complete the following information):**
 - OS: macOS 14.7, Windows 11
 - Device Keyboardio Model 100

**Additional context**
I originally reported this bug [in the Chrysalis repo](https://github.com/keyboardio/Chrysalis/issues/1237), but I now realize it probably belongs here.

Rom1deTroyes, 2024-10-31:
Does your keys on Layer 2 also Shift to Layer 2 ?

If : 
- on Layer 1, you have `e` and `u` both configured as `e` and `u`, with a `Shift to Layer 2`
- on Layer 2, you have `e` configured has `è` and `u` as `ù`

As you can't press `e` and `u` exactly at the same time (one of the keys will be scanned before the other), you will have a sequence like :

- keyPress(e) -> Shift to Level 2
- keyPress(i) -> Press key [ù] (and not a second Shift to Level 2)

or the opposite :

- keyPress(i) -> Shift to Level 2
- keyPress(e) -> Press key [è] (and not a second Shift to Level 2)

Now, in this second case, if you release one of the key :
- keyRelease(i) -> No more Level Shift -> Back to level 1

I agree that the comparison with the behavior of the 2 Shift keys on a legacy keyboard is straightforward and is of course confusing ! 

I don't have my Model100 under the hand to make a full test case, and I'm not sure we can Shift to the same Level we already are (I think there was discussions about that long ago ?), but maybe this can help to find a way to achieve this ?

Happy hacking :-)

mrwonko, 2024-11-01:
> Does your keys on Layer 2 also Shift to Layer 2 ?

They're set to transparent on Layer 2, so they should retain their "Shift to Layer 2" binding from Layer 0.

> on Layer 1, you have `e` and `u` both configured as `e` and `u`, with a `Shift to Layer 2`

I don't understand the "with a `Shift to Layer 2`" part. I have dedicated keys for layer shifting, it's not a side-effect of entering a different button. So staying with the `e` and `u` example, they would both be configured to only `Shift to Layer 2`, without triggering an `e` or `u` press. So I may press one or both of them, and then e.g. `h` `j` `k` `l` change to arrow keys.