keyboardio/Kaleidoscope #1514: Preonic - mouse button clicks not working when connected with Bluetooth

coggro, 2025-07-15:
**Description:** 
Keys assigned to click mouse buttons work when the keyboard is connected via USB cable but do not work when the keyboard is connected via Bluetooth.

**Steps to reproduce the behavior:**
1. Assign keys to click mouse buttons in the configurator. Left, center, and/or right - all are currently bugged.
2. Flash keyboard
3. Connect keyboard via USB cable to confirm keys are assigned correctly. 
4. Test mouse click keys. They should work this way.
5. Disconnect keyboard from USB and reconnect via Bluetooth. 
6. Test mouse click keys. They should do nothing.

**Expected Behavior:**
I expect the left, right, and center mouse click keys to execute their assigned mouse click.

**Environment (please complete the following information):**
 - OS: MacOS
 - Version: 15.5
 - Device: Preonic (Firmware 0.92.7+120) with 2021 MacBook Pro w/ M1 Max

**Additional context**
Attaching current keymap in case it's related to mouse click keys being on shifted layer.

[Chrysalis_Keyboardio-Preonic_layout_2025-07-15_06-58-10.json](https://github.com/user-attachments/files/21237220/Chrysalis_Keyboardio-Preonic_layout_2025-07-15_06-58-10.json)

millipz, 2025-08-03:
I have the same issue using the default out-of-the-box layout for the Preonic

obra, 2025-08-04:
I've been digging into this today. So far, we can see that the events are making it the bluetooth stack to go out. The next things I'm trying are changing our bluetooth mouse report to better match the one Adafruit uses.

obra, 2025-08-07:
https://github.com/espressif/esp-idf/issues/11784 appears to be the same issue (on another platform)

closed 2025-08-09 by obra

coggro, 2025-08-11:
Happy to confirm that this fixed my mouse clicks! Thank you!