keyboardio/Kaleidoscope #1523: Tracking the "flash corruption" issue causing bluetooth reconnects to fail on the preonic

obra, 2025-08-10:
I'd hoped 0.92.8 would resolve this, but it didn't.

https://github.com/meshtastic/Adafruit_nRF52_Arduino/pull/1 is the same issue

coggro, 2025-08-15:
I'm also experiencing this issue. In case it helps, I'm including the full bug report.

**Describe the bug**
Preonic keyboard is failing to reconnect to Bluetooth under varying circumstances, making the cause difficult to pin down.

**To Reproduce**
1. Connect Preonic to computer via Bluetooth.
2. Leave connected for an unknown period of time and/or put computer computer to sleep and/or restart computer, cause is currently unclear.
3. Attempt to use Preonic keyboard or reconnect to Bluetooth. Current result is that it will often fail to reconnect and the computer has to forget the device and/or the keyboard has to be restarted several times.

**Expected behavior**
Regardless of device sleep or extended connection, Bluetooth should easily re-pair to last connected device.

**Environment (please complete the following information):**
 - OS: macOS
 - Version: Sequoia 15.6
 - Device: Keyboardio Preonic (Firmware 0.92.8+124) with 2021 MacBook Pro w/ M1 Max

**Additional context**
n/a

obra, 2025-08-16:
Ok. This is a test build. You can install it by plugging in your preonic while holding in the Hyper key. you'll see a PREONICBOOT drive show up on your desktop. Drag this file onto it.

https://www.dropbox.com/scl/fi/9cr45cfglvsld3ecjhka6/Preonic-202508151955.uf2?rlkey=faj9215f82m4s2mhe7ohadtyb&dl=0

The preonic will install the firmware and then eject the "disk"

After that, run through https://shop.keyboard.io/preonic#recovery-mode

I'd love to know if this improves bluetooth reconnect issues for folks.

obra, 2025-08-17:
> It already "dropped" off without reconnecting already. Recovery with a
> power cycle of the device using the power switvh did resolve it on 1st
> attempt.

@jeroenleenarts Thanks so much for testing this out.
After you powered it off and on, did you need to 'pair' again, or did it
connect automatically after you turned the power switch off and on?

When it "dropped", had the computer gone to sleep?

When it failed to reconnect, did any of the LEDs light up?

Next time that happens, can you test hitting Fun-2 (to switch to bluetooth
slot 2) and then Fun-1 to switch back to the first slot?

hwangy, 2025-08-22:
That's weird, I just had a similar issue as @vrenaville . After adjusting the volume on MacOS (over bluetooth), the keyboard became unresponsive after maybe a minute. Restarting and reconnecting would make it start working again for another minute, but then would again become unresponsive. After seeing @vrenaville , I tried using the rotary button, and that restored functionality.

vrenaville, 2025-08-25:
@obra With the last firmware I don t have the problem actually but as the problem is not every times, I will post it again if the problem occurs again.

Side question, after an upgrade with the Chrysalis, do we still need to make a full recorver every times ?

Thanks for your support,

Vincent

gclimer, 2025-08-25:
Currently on version 0.93.0+128 and experiencing the bluetooth reconnectivity issue.

I performed a factory reset per the recovery steps.

Paired BT#1 with laptop (Fedora 42)

Paired BT#2 with laptop (Windows 11)

Set BT to slot #2.

Reboot Windows 11 laptop. 

Keyboard doesn't re-pair automatically.

Toggle to BT#1, keyboard works on Laptop #1

Toggle back to BT#2, keyboard indicates that it paired to Laptop #2, but laptop does not respond to any keystrokes.

Toggle to BT#1, keyboard pairs and continues to work with Laptop #1.

Prior to updating to 0.93.0+128, I had successfully paired to two laptops - Fedora 42 and Windows 10 - and never experienced an issue on the default firmware from the factory. As soon as I paired the 3rd slot to a Windows 11 laptop, I began to experience issues.

obra, 2025-08-25:
@*gclimer - *With the current firmware, if you try the same setup you had
with the Fedora and Win 10 laptops, how does it behave?

"Paired BT#2 with laptop (Windows 11)

Set BT to slot #2 <https://github.com/keyboardio/Kaleidoscope/issues/2>.

Reboot Windows 11 laptop.

Keyboard doesn't re-pair automatically."

I'm a little confused about that ordering. If the keyboard was already
connected in slot 2, did you toggle away and then back to slot 2?

When the computer was done rebooting and you tried to type on the keyboard,
what did you see on the keyboard's LEDs?

clemensw, 2025-08-30:
With Firmware 0.93.0+128 the Preonic was reliably locking up after more than 10s without a key press, including on newly paired connections.  After locking up, no keystrokes were transmitted and also Fun + 1-5 didn't illuminate the butterfly wings.  The Host continued to see the Preonic as connected and displayed the battery level.  Power cycling the Preonic rapidly restored functionality. (tested on Intel Mac BT1 and iPhone BT3)

Following the "Fully Resetting" steps from https://shop.keyboard.io/pages/preonic-help#recovery-mode resolved the issues and the Preonic is now working reliably as far as I can tell.

clemensw, 2025-08-31:
As an addendum to the above.  After the intel mac was closed overnight, the Preonic could no longer connect the next morning.  None the following got the Preonic to reconect:
- Turning bluetooth off and back on again on the mac
- Switching BT channels on the Preonic
- Powercycling the Preonic
- Powercycling the Mac

The Preonic is not locked up.  It can switch channels and connects to the IPhone on BT3 just fine.  When switching to BT1 (attempting to connect to the intel mac) the #1 Wing briefly turns red, then blue(ish), then tuens off,  then back to blue pulsing twice per second.

gclimer, 2025-09-01:
@obra Are you still tracking this issue as "flash corruption", or something else at this point? I ask because the original PR you linked to has already been closed. It seems we've moved beyond the "flash corruption" issue, and into some other territory.

vrenaville, 2025-09-05:
@obra @hwangy I have open a new issue for the rotary button, I m able to reproduce it now :) https://github.com/keyboardio/Kaleidoscope/issues/1534

jeroenleenarts, 2026-04-19:
@zachgersh I just cloned your repo, build an flashed your BLE branch... Any specific steps you'd like me to take on my Preonic?