keyboardio/Kaleidoscope #1530: When I assign the Ctrl key to the KeyEvent function, it is not generated as a Ctrl key.

asakunotomohiro, 2025-08-28:
In the macroAction function,

**Describe the bug**
The return value of `KeyEvent(KeyAddr::none(), IS_PRESSED, Key_LCtrl);` or `KeyEvent(KeyAddr::none(), IS_PRESSED, Key_RCtrl);` is not the Ctrl key pressed.

I was able to confirm this by pressing the Shift key.
OK ⇒ `KeyEvent(KeyAddr::none(), IS_PRESSED, Key_LShift);` or `KeyEvent(KeyAddr::none(), IS_PRESSED, Key_RShift);`
When the Shift key is passed as an argument, the return value is that the Shift key is pressed.

**To Reproduce**

```
const macro_t *macroAction(uint8_t macro_id, KeyEvent &event) {
KeyEvent eventLShift = KeyEvent(KeyAddr::none(), IS_PRESSED, Key_LShift);
KeyEvent eventRShift = KeyEvent(KeyAddr::none(), IS_PRESSED, Key_RShift);
  switch (macro_id) {
  case CtrlKeyPush:
    if(!event.key.isKeyboardKey()) {
	for (Key key : kaleidoscope::live_keys.all()) {
		if (key == eventLShift.key || key == eventRShift.key
			) {
Macros.type(PSTR(“eventCtrlKey()"));
		}
	}
    }
  break;
```
When the Shift key and the assigned key for the macro function were pressed, the “eventCtrlKey()" string was output.

If the Shift key argument was changed to the Ctrl key, the string was not output.

The isKeyboardModifier() method also did not recognize the Ctrl key being pressed.
The isMomentary() method also did not recognize the Ctrl key being pressed.

**Expected behavior**
A clear and concise description of what you expected to happen.
I wasted time because I didn't realize that pressing the Ctrl key was the only option not supported.

**Environment (please complete the following information):**
 - OS: macOS Sequoia(Apple M1 Max)
 - Version 15.3.1（24D70）
 - Device [The Keyboardio Preonic](https://www.kickstarter.com/projects/keyboardio/preonic)

**Additional context**
Only the Ctrl key is treated specially.

gedankenexperimenter, 2025-09-10:
If you share your whole sketch, I'll look into it when I have the time.

asakunotomohiro, 2025-10-05:
The product description said that it would be possible to create custom layouts.
However, there are currently bugs and it still cannot be used as a keyboard.
When will it be completed?
When can I get the finished product?
I would like to know when it will be completed.