keyboardio/Kaleidoscope #1532: 【Qukeys】The CTL_T key is not effective and the argument key is used instead.

asakunotomohiro, 2025-09-01:
**Describe the bug**
I set the **CTL_T(Enter)** key, but the Ctrl key did not work.
Also, when I changed CTL_T to GUI_T, it worked as a GUI key.

Also, even if you set QUKEYS individually, it will not be effective.

**To Reproduce**
Set the `CTL_T(Enter)` key to the layer.

Also,
```
	QUKEYS(
		kaleidoscope::plugin::Qukey(0, KeyAddr(3, 0), Key_LeftControl),	// Enter(Left side of A key)
		kaleidoscope::plugin::Qukey(0, KeyAddr(4, 1), Key_LeftGui)	// Z
	);
```

**Expected behavior**
I thought that holding the "CTL_T(Enter)" key would make it function as the Ctrl key.
also, Tapping it functions as the Enter key.

**Environment (please complete the following information):**
 - OS: macOS Sequoia(Apple M1 Max)
 - Version 15.3.1（24D70）
 - Device [The Keyboardio Preonic](https://www.kickstarter.com/projects/keyboardio/preonic)

**Additional context**
There were other cases where the Ctrl key did not work.
https://github.com/keyboardio/Kaleidoscope/issues/1530

vrenaville, 2025-09-05:
I second that, some keys work some not, like arrow up / arrow down does not work, but arrow lef and arrow right seems to work

vrenaville, 2025-09-19:
@asakunotomohiro did you have tried rightalt, it seems also impossible to use this key

Vincent

asakunotomohiro, 2025-10-05:
The product description said that it would be possible to create custom layouts.
However, there are currently bugs and it still cannot be used as a keyboard.
When will it be completed?
When can I get the finished product?
I would like to know when it will be completed.

My hope is that this takes priority over the Unicode issue.